jlecomte/ascom-telescope-cover #1: Merge some of your projects

JeffStor, 2023-03-12:
Hello Julien,

No issue, no pull request, just a thought. My name is Jean-Francois, I live in France, and I am a (low level) astrophotographer.
I use NINA and Ascom devices. My setup is EQ6-R / Askar FRA600 /ASI2600color.

I was seeking for an automated dust cap for my telescope and I found your videos ! (Dark Sky Geek)
You did such a great job for 2 projects in which I am interested : dust cap and focuser. 
I am not a developer and I discovered your channel today !
I bought hardware stepper-motors drivers called TIC from Pololu and I did a small GUI to drive manually stepper-motors.
This was easy, but to integate TIC with NINA (native or Ascom) is much to dffificult for me.

A stepper-motor and a stepper motor is much less expensive that any Zwo, Pegasus or else equipment.

If for any reason, you plan to use Pololu TIC hardware to drive stepper-motors for your new projects, please let me know. 

Sincerely

Jean-Francois
jf.eble@laposte.net

jlecomte, 2023-03-12:
Thank you for your interest. I do not plan to use "Pololu TIC" hardware (I have never heard of that brand before). I generally stick to dirt cheap, standard stepper motors or servos that you can find anywhere online. That being said, adapting what I've done to your specific hardware shouldn't be too difficult. Best of luck!

closed 2023-03-12 by jlecomte

JeffStor, 2023-03-12:
Thank you for your quick answer. By the way I choosed TIC + stepper-motors because I can buy both on Amazon for around  50 euros (plus a 3D printed hardware for virtually nothing) when a focuser from ZWO or Pegasus costs more than  250 euros

Clear skies !