jlecomte/ascom-telescope-cover #2: Issue with FreeCAD model

rlavriv, 2023-05-01:
When I am trying to change globals in FreeCAD model to adjust to my OTA, I get issue with model recompile.

The issue seem to be with USB Connector hole sketch, which shows in a wrong plane:
[image]

Any ideas how to fix it?

jlecomte, 2023-05-01:
That is because of the topological naming problem, which is a huge issue in FreeCAD. See:

* https://wiki.freecad.org/Topological_naming_problem
*  https://ondsel.com/blog/freecad-topological-naming/

Unfortunately, I did not follow the best practices when designing this model, and therefore it is a victim of this issue.

Since I may upgrade my scope to a larger one, I may completely redesign this in FreeCAD, but this time doing it correctly (and also simplifying the design a bit)...

rlavriv, 2023-05-01:
Is there absolutely no way to salvage it?

jlecomte, 2024-05-25:
I don't think so. Now, I'm hearing that FreeCAD 0.22 will have mostly solved the topological naming issue, so it might be worth trying then. best of luck!