jlecomte/ascom-telescope-cover #6: Thanks for this project

astroalex80, 2023-10-08:
Dear Julien,

Thanks for making the cover as well as flat panel available.

Converted the cad files to Fusion360 to make them fit my scope. Combined both flat panel and cover functionality.

Honesty my mistake of not having used an SAMD21 / Seeeduino XIAO before. What a powerful tiny piece of electronics.

Instead the usual servo lib I'm using one called slowServo allowing good control of the speed and it is working asynchronously and not blocking when servo moves. It therewith better follows the ASCOM specs like reporting status when moving.

The fact of fast closure move on power-up when cover is not closed I could overcome by writing either panel open, closed or moving to the Flash memory of the XIAO. Therewith current cover status remains on power up, unless the XIAO is disconnected while servo is moving. In this case the cover opens on the re-connection, as feels more save to me before moving fast into the dew shield.

3d parts are printed, cover is moving but wiring is still on a breadboard. Will assemble it next days and happily share some pictures.

Kind regards from Germany,
Alex

jlecomte, 2023-10-09:
Thanks for the kind words, Alex! These are really good suggestions, and I will incorporate some of them into the version 2 of this project, next year. Among other things, I want the cover to know which state it is in when powered up, and this can be accomplished using a potentiometer and a bit of logic in the software. Tschüss!

closed 2023-10-09 by jlecomte

astroalex80, 2023-10-10:
By the way regarding reading the cover position. I havent been aware they exist but just ordered two of these:
Feetech FB5118M Digital Servo 7.4V 17kg.cm with Position Feedback

There is a 4. wire linked to the internal potentiometer of the servo and therewith having the build-in capability of providing postion.

jlecomte, 2023-10-10:
I did not know that some servos had position feedback. I just ordered one on Amazon, I’ll be experimenting with it. If it works, this would be a brilliant solution! Thanks a lot :)