jlecomte/ascom-telescope-cover #9: Cannot connect properly in NINA

sidiouss, 2024-05-24:
Hello,
I've successfully prepared slightly modified flat panel and telescope cover projects which I am very thankful for. I managed to register both drivers and setup properly the flat panel in Nina. However, I cannot launch the cover in the Nina software. It says both that the switch is connected, but then also prompts that there is invalid response from the cover. Can you help me investigating this issue?
The cover properly shows up in the device manager as a com port, I can send the commands through the COM terminal so I know the cover works, but I cannot launch it with NINA. I am using the newest dll and firmware is uploaded to Arduino Nano.

Prompt from NINA:
[image]

jlecomte, 2024-05-24:
I do not know what is causing this, and I cannot provide you with any support.

closed 2024-05-24 by jlecomte

sidiouss, 2024-05-25:
Hi,
You may not be eager to solve such problems and I understand that. However, I have run conformance checker on your driver and as you are more familiar with ASCOM platform, maybe you could identify the underlying issue presented in the following report:

 
ConformanceCheck ASCOM Device Conformance Checker Version 6.6.8048.17861, Build time: 2022-01-13 08:55:24
ConformanceCheck Running on: ASCOM Platform 6.6 SP2 6.6.2.4294
 
ConformanceCheck Driver ProgID: ASCOM.DarkSkyGeek.Switch
 
Error handling 
Error number for "Not Implemented" is: 80040400
Error number for "Invalid Value 1" is: 80040405
Error number for "Value Not Set 1" is: 80040402
Error number for "Value Not Set 2" is: 80040403
Error messages will not be interpreted to infer state.
 
21:45:03.873 Driver Access Checks              OK       
21:45:04.535 AccessChecks                      OK       Successfully created driver using late binding
21:45:06.583 AccessChecks                      OK       Successfully connected using late binding
21:45:06.588 AccessChecks                      INFO     The driver is a .NET object
21:45:06.593 AccessChecks                      INFO     The AssemblyQualifiedName is: ASCOM.DarkSkyGeek.Switch, ASCOM.DarkSkyGeek.TelescopeCover, Version=1.0.0.0,
21:45:06.599 AccessChecks                      INFO     The driver implements interface: ASCOM.DeviceInterface.ISwitchV2
21:45:07.277 AccessChecks                      INFO     Device does not expose interface ISwitch
21:45:09.339 AccessChecks                      INFO     Device exposes interface ISwitchV2
21:45:10.014 AccessChecks                      OK       Successfully created driver using driver access toolkit
21:45:12.063 AccessChecks                      OK       Successfully connected using driver access toolkit
 
Conform is using ASCOM.DriverAccess.Switch to get a Switch object
21:45:12.099 ConformanceCheck                  OK       Driver instance created successfully
21:45:14.211 ConformanceCheck                  OK       Connected OK
 
Common Driver Methods 
21:45:14.265 InterfaceVersion                  OK       2
21:45:14.305 Connected                         OK       True
21:45:14.352 Description                       OK       DarkSkyGeek’s Telescope Cover
21:45:14.398 DriverInfo                        OK       DarkSkyGeek’s Telescope Cover ASCOM Driver Version 1.0
21:45:14.445 DriverVersion                     OK       1.0
21:45:14.492 Name                              OK       DarkSkyGeek’s Telescope Cover
21:45:14.538 CommandString                     INFO     Conform cannot test the CommandString method
21:45:14.547 CommandBlind                      INFO     Conform cannot test the CommandBlind method
21:45:14.557 CommandBool                       INFO     Conform cannot test the CommandBool method
21:45:14.565 Action                            INFO     Conform cannot test the Action method
21:45:14.574 SupportedActions                  OK       Driver returned an empty action list
 
Properties 
21:45:14.734 MaxSwitch                         OK       1
 
Methods 
21:45:15.001 SwitchNumber                      ISSUE    Received ASCOM.DriverException(0x80040401), please use ASCOM.InvalidValueException to report invalid values
21:45:15.009 SwitchNumber                      ERROR    Unexpected DriverException(0x80040401), when a switch ID below 0 was used in method: CanWrite: CanWrite set - '-1' is an invalid value. The valid range is: 0 to 0.
21:45:15.019 SwitchNumber                      ISSUE    Received ASCOM.DriverException(0x80040401), please use ASCOM.InvalidValueException to report invalid values
21:45:15.027 SwitchNumber                      ERROR    Unexpected DriverException(0x80040401), when a switch ID above MaxSwitch was used in method: CanWrite: CanWrite set - '2' is an invalid value. The valid range is: 0 to 0.
21:45:15.037 SwitchNumber                      ISSUE    Received ASCOM.DriverException(0x80040401), please use ASCOM.InvalidValueException to report invalid values
21:45:15.045 SwitchNumber                      ERROR    Unexpected DriverException(0x80040401), when a switch ID below 0 was used in method: GetSwitch: GetSwitch set - '-1' is an invalid value. The valid range is: 0 to 0.
21:45:15.055 SwitchNumber                      ISSUE    Received ASCOM.DriverException(0x80040401), please use ASCOM.InvalidValueException to report invalid values
21:45:15.063 SwitchNumber                      ERROR    Unexpected DriverException(0x80040401), when a switch ID above MaxSwitch was used in method: GetSwitch: GetSwitch set - '2' is an invalid value. The valid range is: 0 to 0.
21:45:15.075 SwitchNumber                      ISSUE    Received ASCOM.DriverException(0x80040401), please use ASCOM.InvalidValueException to report invalid values
21:45:15.085 SwitchNumber                      ERROR    Unexpected DriverException(0x80040401), when a switch ID below 0 was used in method: GetSwitchDescription: GetSwitchDescription set - '-1' is an invalid value. The valid range is: 0 to 0.
21:45:15.095 SwitchNumber                      ISSUE    Received ASCOM.DriverException(0x80040401), please use ASCOM.InvalidValueException to report invalid values
21:45:15.106 SwitchNumber                      ERROR    Unexpected DriverException(0x80040401), when a switch ID above MaxSwitch was used in method: GetSwitchDescription: GetSwitchDescription set - '2' is an invalid value. The valid range is: 0 to 0.
[... cut ...]
21:45:39.739 SetSwitchValue                    ERROR    Unexpected DriverException(0x80040401), when setting a value above SwitchMaximum was set: 2: SetSwitchValue set - '2' is an invalid value. The valid range is: Switch(0) range 0 to 1.
21:45:42.820 SetSwitchValue                    WARNING  Skipping multi state tests because of earlier errors
21:45:42.837 SetSwitchValue                    OK         Switch has been reset to its original state
 
 
Conformance test complete 
 
Your driver had 22 errors, 1 warning and 23 issue