jlecomte/ascom-telescope-cover #10: Problems in WIN11 64bit

thegoodpope, 2024-05-25:
This terrific complexity with Window versions, 32/64 bit, ASCOM versions, .NET frameworks will kill me...
[image]

thegoodpope, 2024-05-25:
Registering in 32 bit ascom is ok, but when I tyy to connect in NINA
[image]

jlecomte, 2024-05-25:
I have had no such problem. I'm not sure what you're doing, but the Visual Studio project should be configured properly out of the box. When I last published the DLL, I simply opened the project, built it, and copied the DLL.

thegoodpope, 2024-05-26:
I've done the same, but got problems with regasm as i've alteady written.
This last thing has not to do with recompilation: I used your original
compiled DLL as downloaded from the resources sectiin. In my old PC It
worked flawlessy, buy when i've tried to install It on the win11 PC that
controls mount and camera (remotely) It has occurred the strange
situation....