jlecomte/ascom-telescope-cover #11: Adjusting servo end courses and speed (delay)

vef445, 2024-06-17:
Hi,
First of all, be aware that I'm in no way a professional developer :)

With the driver you built, I wasn't able to get the full course of my 270deg. servo. A quick investigation revealed that the default values provided by Servo.h when attaching the servo (min pulse width = 544 and max pulse width = 2400) were not ideal for my setup. (Using the cheap servos from Amazon that you listed).

Instead of just editing the Arduino code with alternative (and fixed) values, I decided to add 'pulse min' and 'pulse max' as part of the Ascom driver profile.
[image]
While I was there, I also added the servo write "delay" as a parameter too. (the default value 30 isn't the best for my setup)

I have no idea how I can properly share my code with you or if you're even interested. I'm simply attaching a ZIP here that contains both the driver and the Arduino file. Let me know if there is a better method.

Comments/critics very welcome!

-Guillaume
[Motorized_cover_gvi.zip](https://github.com/user-attachments/files/15864649/Motorized_cover_gvi.zip)

jlecomte, 2024-06-17:
Hello! Great work! Note that this project is now on life support so I will not make any changes to it. However, I have been working on a new version of this project, which brings a lot of improvements, and fixes the servo pulse min/max thing you mentioned. I do not have an ETA to release that project. Thanks!

closed 2024-06-17 by jlecomte

vef445, 2024-06-17:
> Hello! Great work! Note that this project is now on life support so I will not make any changes to it. However, I have been working on a new version of this project, which brings a lot of [...]

I’ll keep an eye on what’s coming next!